kiswitch/kiswitch #1: Where can I get the 3d models?

geocine, 2020-09-25:
Where is this located? `Switch_Keyboard_Kailh.3dshapes`. Are you willing to share it?

perigoso, 2020-09-25:
At the moment I don't have the 3d models included, they are linking to the location of the 3d models of the official libraries, but the official libraries don't include it right now either, i can't get you the model for cherry mx switches quickly, but i don't know that I've seen 3d models for the hotswap sockets

geocine, 2020-09-27:
I am specifically looking for a WRL 3d model of a kailh hotswap socket, with textures included. What did you mean by this?
> i don't know that I've seen 3d models for the hotswap sockets
I actually found that model being referenced on your footprints

perigoso, 2020-09-28:
I meant that I hadn't seen it modeled anywhere so I assumed it would have to be modeled from scratch, but I've found it now too.

perigoso, 2020-09-29:
@geocine 3D models are now included as part of the library in [here](https://github.com/perigoso/Switch_Keyboard/tree/main/packages3d).

closed 2020-09-29 by perigoso