kiswitch/kiswitch #4: Pad numbers for SW_Cherry_MX and SW_Hybrid_Cherry_MX_Alps

adamws, 2021-02-09:
Due to difference in pad numbering between MX and Hybrid switches, footprint replacement is impossible.
I think that pad 1 for hybrid switch should be on the left as well.
[image]
[image]

perigoso, 2021-02-10:
I did this because of conflicting documentation I believe, where the manufacturer stated what they considered to be pin 1, I adopted it, resulting in conflicts, in hindsight i should've just chosen a numbering and used it everywhere, I shall change it when I have time, thank you for taking the time to open an issue.

perigoso, 2021-02-17:
Corrected

closed 2021-02-17 by perigoso