kiswitch/kiswitch #5: ISO enters

adamws, 2021-02-13:
- both Alps_Matias and Hybrid provides single orientation (which needs to be rotated by user in 99.9% cases) and Cherry_MX provides rotated variants
- I think, that rotated variants for Cherry_MX are not needed. I'm aware that for some keyboards (for example pok3r) ISO enter switch is rotated but as far as I know it isn't considered 'cherry standard'. (is ISO switch rotation mentioned in any documentation?)

    In fact, I'm under impression that most community keyboards are using ISO enters in 'normal' orientation (by 'normal' I mean same as every other switch) hence my proposition is to have single ISO footprint in all libraries (Cherry/Alps and Hybrid) like this:
[image]

rare occasions when someone wants rotated mounting holes for ISO enter switch can be handled with 'base' footprints that each library here have (after all, all derivatives differ only with key cap outline right?)

if you insist to keep rotated variants, perhaps it would be best if Alps/Hybrid also have them.

perigoso, 2021-02-15:
The reason alps and hybrid don't have rotated footprints is because unlike the cherry mx standard, the mounting on alps is not symmetrical, meaning with standard keycaps you would not be able to mount the switch "sideways".

But you're right, I think I makes sense to remove them, it's very niche, a "normal" orientation and a flipped is probably enough for 99% of applications and that's what the base is there for.

perigoso, 2021-02-17:
Decided to keep the extra rotations on cherry MX only, but changed the default orientation like you suggested.

closed 2021-02-17 by perigoso