kiswitch/kiswitch #7: .lib files?

GandalftheGay, 2021-04-27:
am i just dumb (pretty new to this) or how do i import and use them without the .lib file?
i can open the kicad-mod files in the "edit footprints tool" but what do i do from there? cant find them in the eeschema UI

perigoso, 2021-04-27:
this library does not provide `.lib` files, aka symbols, only footprints.

The way this is meant to work is you use a generic switch symbol from the official Kicad libraries, and afterwards assign it the correct footprint from this library.

GandalftheGay, 2021-04-28:
Didnt expect such a fast response, thanks for the info!

can i ask you another question?
if i want to put multiple, for example 1u keys, in a row/colum.. do i have to space them or can i line them up directly next to each other with the outlines of the footprint touching?

perigoso, 2021-04-29:
You can line them up with the user drawings that com with the footprints, i believe the lines overlap, but you can confirm by putting the on top of each other and moving one by 19.05mm

GandalftheGay, 2021-05-05:
sorry for the late reply again and thanks for your help.
could  you directly link to the stabilizer package in your readme-"stablizers" description?
its otherwise kinda hard to find if you are a github noob like me.
sorry for all my dumb questions, im really glad you are replying so thanks again!

perigoso, 2021-05-05:
sorry, I'm not following the question? when I say package I really mean footprint library, It is in the same repo [right here](https://github.com/perigoso/keyswitch-kicad-library/tree/main/modules/Mounting_Keyboard_Stabilizer.pretty)
You're supposed to put your regular switch footprints on the pcb, and add the stabilizer footprints on top of the switches manually

GandalftheGay, 2021-05-05:
sorry for my poor description of the problem but your answer answered all my questions.
i overlooked the seperate stabilizer folder in the repo. thanks for the help and the great repo!

closed 2021-05-05 by GandalftheGay