kiswitch/kiswitch #8: Switch cutouts?

cgarcia2097, 2021-06-21:
Is it also possible to include switch cutouts in this library, for those interested in rendering one's plates in KiCAD? (IIRC it also supports exporting Edge.Cuts into SVG)

perigoso, 2021-06-21:
you mean a new footprint with just the cutout outline? its possible sure, just a quick script

obra, 2021-12-10:
I think that it would be most sensible to do this by picking one of the User layers as the 'plate cutout' layer on every switch footprint.

perigoso, 2021-12-10:
That is my latest idea, have it on a Eco1 or something, need to find the most switable. this allows the export of a plate dxf directly for a kicad keyboard, which is pretty neat

kbrock, 2021-12-12:
I'm surprised how

perigoso, 2021-12-24:
as of now all foot prints that i think are supposed to have cutouts have one on Eco1.User

If any are missing please open a new issue

closed 2021-12-24 by perigoso