kiswitch/kiswitch #9: HOW TO INSERT THIS LIBRARY INTO KICAD

Davides98, 2021-08-16:
I'm new into kicad and I did not understand how to insert this library.

perigoso, 2021-08-16:
Have you followed the instructions in the Readme?

Davides98, 2021-08-17:
Yes, I try to insert into preferences --> manage symbol 
pretty files but I can't

perigoso, 2021-08-17:
This library does not have any symbols, only footprints, different menus

Davides98, 2021-08-18:
Okay,
I try it.

Footprint Editor -> Footprint Libraries

i didn't understand which folder i need to link inside footprint libraries

Sorry but I'm new into this software

perigoso, 2021-08-19:
the folders inside `modules/` are the ones you want to select
`Mounting_Keyboard_Stabilizer.pretty`
`Switch_Keyboard_Alps_Matias.pretty`
`Switch_Keyboard_Cherry_MX.pretty`
`Switch_Keyboard_Hybrid.pretty`
`Switch_Keyboard_Kailh.pretty`

closed 2021-10-18 by perigoso