kiswitch/kiswitch #22: workflow to check that the footprints have been regenerated

perigoso, 2021-12-23:
-  setup a workflow to check that the footprints have been regenerated when scripts have changed
-  pre-commit of the above if possible

Lotier, 2022-01-22:
I tried messing around with this a bit. I can easily generate all the footprint files. It doesn't seem to hard to create the STP files, but I can't seem to figure out how you make the WRL files. But I can then spin up a way that on every commit to `main` would run a workflow to create the zip from all the source files again. So how do you make the WRL files?

perigoso, 2022-01-22:
Right now 3d files are generated manually

Lotier, 2022-01-25:
Manually from within FreeCAD?

perigoso, 2022-01-25:
yes

Gigahawk, 2022-05-04:
I've written an Action to export FreeCAD documents at https://github.com/Gigahawk/freecad-export-action.

You can see a demo of it at https://github.com/Gigahawk/freecad-export-action-test/actions/runs/2269007904

We can generate files automatically, but there are a few implementation details I haven't worked out:
- Slight differences in environment (version of FreeCAD and its dependencies, time of export) will cause the exported files to be different. This prevents us from checking if the source and export match by simply rerunning the export.
- Setting up a hook to trigger exports locally would require FreeCADCmd to be on the path, or some elaborate way of detecting it would have to be built.

There are a couple of ways to work around this:
    - An export action that just monitors for changes to the source files on master and will automatically push a new commit with the correct exported files
    - Provide exported files as a download, users would be expected to download them from the action and include them in the PR before they get merged