kiswitch/kiswitch #23: Add led locations

perigoso, 2021-12-24:
maybe a led symbol within a square that shows the location

some footprints already had marking that were removed in the process of being scripted like the ones in #11 #12

AlmoGsh98, 2021-12-24:
hey!
I would like to help you with this issue.
this is my first time contributing.
can you explain more about the issue? where do you want to add the led locations?

perigoso, 2021-12-24:
I want to add some sort of marking in the User.Drawings like the keycaps are, i was thinking a rectangle representing the area available for the leds, with maybe a center marker and and led symbol to make it clear to end user

To implement i was thinking of making a Node class like did with the keycaps, the constructor would take in the size of the area and it's center location, and the switch classes would import that directly and append to self

Honestly the most work will be gathering the location and size info of the led location for each switch, and if it exists in the first place

perigoso, 2022-05-26:
> I changed my mind on how I want this to work, I want to have the LEDs embedded in the footprint, behaving like the keycaps do right now in the scripts, they wont be included in the default library but will be generate-able now manually, and later through the TODO plugin
> This comes with a couple problems to solve, the positioning needs to be handled somehow in the switch class, presumably after the Led class has been created
> And the hardest I think is handling the Library symbols, perhaps for now we could have a couple fixed options, essentially one for each LED, but at some point automatic generation and assignment should be considered
from #40