kiswitch/kiswitch #25: Fix hotswap choc outline

perigoso, 2021-12-24:
The current outline used in hotswap choc switches is far for it's real mechanical outline, needs fixing

perigoso, 2022-05-07:
note: there is a fork that has a pretty good alternative, consider merging.

rrotter, 2022-05-13:
Thanks for the shoutout. I made a lot of incompatible changes in my fork. I'll cherry-pick the outline changes and send a PR.

perigoso, 2022-05-13:
That would be great!

rrotter, 2022-05-19:
Still working on this. I have a new outline that is much better, but it uses `fp_arcs` which aren't supported by PolygoneLine, and they are KiCad 6 `fp_arcs`, which aren't even supported by https://gitlab.com/perigoso/kicad-footprint-generator/.

perigoso, 2022-05-19:
hm, yeah the footprint generator is still somewhat stuck on kicad 5

closed 2022-05-20 by perigoso