kiswitch/kiswitch #26: Switch request: choc v2

perigoso, 2021-12-24:
**What's the name and manufacturer if any of this switch?**

Kailh Choc V2 `p/n`

**Is there any equivalent switch that you know of?**

N

**What keycap size and orientations are available for this switch?**

Same as choc v1?

**What is the standard spacing for the keycaps?**

Same as choc v1?

**Provide any documentation you have about this switch.**

`link to site`

obra, 2021-12-31:
I thought I'd provided this one, but maybe I didn't?

perigoso, 2021-12-31:
You did not, you provided kailh kh, and choc mini, which is different I'm pretty sure?

obra, 2021-12-31:
Yup. I’ll need to dig out my footprint for the CPG 1353

obra, 2021-12-31:
I apologize for not doing this as a PR, but I'm running for a plane.

Notes: CPG1353 and CPG1350 are VERY similar. The 1353 (Choc V2) needs a larger center cutout and has a single off-center fixing peg, not two side-mounted fixing pegs like the Choc V1.

I'd recommend defaulting to a combined Choc V1 and V2 footprint with the larger (5mm) center hole and all three fixing holes.  The only other complication is that the LED centerpoint on the two switches is a tiny bit different.

I'd also recommend that these changes be made to the Choc Socket footprint.

```
(footprint "Kailh-CPG1353" (version 20211014) (generator pcbnew)
  (layer "F.Cu")
  (tedit 61CF52C3)
  (attr smd)
  (fp_text reference "REF**" (at -1.016 3.048) (layer "F.SilkS")
    (effects (font (size 1 1) (thickness 0.15)))
    (tstamp c5ef60fa-08b4-4038-80b2-809ee9c343bc)
  )
  (fp_text value "lp" (at 2.54 4.826) (layer "F.Fab")
    (effects (font (size 1 1) (thickness 0.15)))
    (tstamp 05963e4f-8a19-41a2-8c10-33c354d0ea8e)
  )
  (fp_line (start 7 -6) (end 7 -7) (layer "B.SilkS") (width 0.1) (tstamp 0f04d8dc-e9cd-4a92-829d-80b1e24fb391))
  (fp_line (start -7 -7) (end -7 -6) (layer "B.SilkS") (width 0.1) (tstamp 416b4805-e2b9-4bb9-a448-f8adb8428678))
  (fp_line (start -6 -7) (end -7 -7) (layer "B.SilkS") (width 0.1) (tstamp 9d59aaa7-683b-44fd-962c-e4d71765be5c))
  (fp_line (start 7 7) (end 7 6) (layer "B.SilkS") (width 0.1) (tstamp aa69b52d-8d3b-4ade-bb26-1dff80416d12))
  (fp_line (start -7 7) (end -6 7) (layer "B.SilkS") (width 0.1) (tstamp c4f1985a-6264-4810-912f-fbfc96a3b1d7))
  (fp_line (start -7 6) (end -7 7) (layer "B.SilkS") (width 0.1) (tstamp c8fdbe45-a6a5-4745-acd6-5cfdccb29764))
  (fp_line (start 7 -7) (end 6 -7) (layer "B.SilkS") (width 0.1) (tstamp c9e5069d-5996-4dbe-ba84-79f77eb5ab91))
  (fp_line (start 6 7) (end 7 7) (layer "B.SilkS") (width 0.1) (tstamp ca710591-a2b8-4ad9-8342-193c67aea6e5))
  (fp_circle (center 0 -4.93) (end 1.30384 -4.93) (layer "B.SilkS") (width 0.15) (fill none) (tstamp c0c1a07b-2241-4bf0-b1a9-d9e099e08c99))
  (fp_line (start -7 -7) (end 7 -7) (layer "Dwgs.User") (width 0.12) (tstamp 59652656-c051-44fe-9c14-7317f91ef1ba))
  (fp_line (start 7 -7) (end 7 7) (layer "Dwgs.User") (width 0.12) (tstamp 852ab9b7-d9ea-44ed-af37-1b96f9feda0a))
  (fp_line (start -7 7) (end -7 -7) (layer "Dwgs.User") (width 0.12) (tstamp 9b3cd9ad-f4b5-4583-967f-5c0f61fb3cd0))
  (fp_line (start 7 7) (end -7 7) (layer "Dwgs.User") (width 0.12) (tstamp aa68e1eb-812d-4c8a-8e4a-254972f3b66a))
  (pad "" thru_hole circle (at -5 -5.15) (size 1.8 1.8) (drill 1.6) (layers *.Cu *.Mask) (tstamp 1a2187bb-0281-4ca6-931f-b647838a6c70))
  (pad "" np_thru_hole circle (at 0 0) (size 3.4 3.4) (drill 5) (layers F&B.Cu *.Mask) (tstamp e42b7c69-5768-4568-9556-43e81993fa96))
  (pad "1" thru_hole circle (at 5 3.8 180) (size 1.5 1.5) (drill 1.2) (layers *.Cu *.Mask) (tstamp 04b0b95f-8c2f-4f37-b712-2393b100dbe5))
  (pad "2" thru_hole circle (at 0 5.9 180) (size 1.5 1.5) (drill 1.2) (layers *.Cu *.Mask) (tstamp 53ed5f01-56ef-4878-bfee-8f83856b3166))
)
```

obra, 2021-12-31:
Choc v1:

[image]

Choc v2:
[image]

perigoso, 2022-01-03:
hm, my current footprint doesnt seem to match either one, I'm conviced I got this from a datasheet though.
[image]

perigoso, 2022-01-03:
Looks like the existing one is supposed to be compatible with both already, but the location of the bottom locator pin is not quite right according to that doc, also is that bottom pin supposed to be soldered too?

If i got this right:
V2:
[image]
V1:
[image]

obra, 2022-01-03:
The stabilization pin on the v2 switch is made of copper and intended to be soldered in place (if you are not using hot-swap sockets)

I'll lasercut some samples of the footprints later to test out the sizes of everything, but I think your existing choc footprint should be good already.

perigoso, 2022-01-12:
did you ever do that print?

Kriechi, 2022-01-16:
@obra I think you mixed v1 and v2 in your comment above https://github.com/perigoso/keyswitch-kicad-library/issues/26#issuecomment-1003441580

Your footprint looks like it was designed for both versions, while the reference name in the footprint says explicitly v1.

I think the extra stabilizer pin on v2 is a bit off in your footprint, according to the datasheet it should be 5mm and 5.15mm from the center.  

The NPTH is also too small, it should be 1.6mm instead of 1.3mm.

perigoso, 2022-01-16:
Thanks @Kriechi that's what I thought

obra, 2022-01-18:
Sorry, I'm running pretty behind on projects.

@Kriechi 

Thank you. That's correct. I've corrected my post.

> @obra I think you mixed v1 and v2 in your comment above [#26 (comment)](https://github.com/perigoso/keyswitch-kicad-library/issues/26#issuecomment-1003441580)

> I think the extra stabilizer pin on v2 is a bit off in your footprint, according to the datasheet it should be 5mm and 5.15mm from the center.
> 
> The NPTH is also too small, it should be 1.6mm instead of 1.3mm.

I agree with that.

[image]

My recommendations: 

Center hole should become 5.05mm

v2 fixing pin should move as @Kriechi specifies.

perigoso, 2022-01-18:
Dont worry, no pressure here.

Thank you so much for testing things!

thatguyisjames, 2022-03-09:
I just ran into this issue as it seems no one has a public footprint to fit V1 and V2. I made the changes @obra talked about to your files. I just want to confirm that the laser cut photo has the 3rd fixing pin at X -5, Y 5.15. My photo editing skills seem to say that is what measurements you used for the laser cut. It is implied by the conversation, but I do not want to have my second batch of boards need the fixing pins cut again.
[image]

perigoso, 2022-03-09:
@thatguyisjames
these are the correct dimensions, sorry you had trouble with this, I haven't had the time to fix it yet
https://github.com/perigoso/keyswitch-kicad-library/issues/26#issuecomment-1003441580

obra, 2022-03-09:
> I just ran into this issue as it seems no one has a public footprint to fit V1 and V2. I made the changes @obra <https://github.com/obra> talked about to your files. I just want to confirm that the laser cut photo has the 3rd fixing pin at X -5, Y 5.15. My photo editing skills seem to say that is what measurements you used for the laser cut. It is implied by the conversation, but I do not want to have my second batch of boards need the fixing pins cut again. V1V2 Size <https://user-images.githubusercontent.com/17017012/157490949-eb7c201c-d692-4f45-854e-63234f341559.PNG>

To make extra-special-sure, one thing you can do is to export your board to an SVG and cut it on the laser before sending it to fab.

thatguyisjames, 2022-03-10:
No need for apologies, it was my fault for not measuring twice and cutting once.

As I don't have a laser cutter, I guess I could 3d print these holes to test fit. But for future proof other visitors.

All I did was change the center hole to 5.05 and move the 3rd fixing pin to the measurements I listed above (hole size 1.6, copper 2.6, location X -5, Y 5.15). It will be a few months before I have a board back to confirm this on PCB fitment, so take that with a grain of salt.

Thanks all for the help and quick replies.

perigoso, 2022-03-14:
4a6e54881377de90c06b4d26fc5cf1e707c7aed8 should fix the bad dimensions, I made a bugfix release for this [v2.1.1](https://github.com/perigoso/keyswitch-kicad-library/releases/tag/v2.1.1)

perigoso, 2022-03-14:
ups. I forgot the hotswaps.

perigoso, 2022-03-14:
06174cae697d23f414e05371e73ce802a57431cf second fix, bugfix release: [v2.1.2](https://github.com/perigoso/keyswitch-kicad-library/releases/tag/v2.1.2)

perigoso, 2022-03-14:
@obra if you could validate this I would greatly appreciate it! no pressure though, I understand time is limited.

obra, 2022-03-14:
@perigoso That fixing pin placement does match the datasheet, as I read it.

perigoso, 2022-03-14:
Unified things and now we are able to generate switches for V1, V2 or both

What the footprints currently look like

These changes are currently in PR #37

[image]
[image]
[image]
[image]
[image]

closed 2022-03-17 by perigoso