kiswitch/kiswitch #56: v2.3 is not usable with autorouters due to keepouts in dsn file

adamws, 2023-01-08:
Hi, this library (in version 2.3) does not work with auto-routers programs (for example https://github.com/freerouting/freerouting/).
In exported dsn file of PCB I see that each switch has two keepouts defined in place of through-hole pads, which is preventing routers to use those pads.

Version 2.2 does work. Here is comparison of related dsn part:

version 2.2 | version 2.3
--- | ---
(keepout "" (circle F.Cu 4000))<br>(keepout "" (circle B.Cu 4000))<br>(keepout "" (circle F.Cu 1750 -5080 0))<br>(keepout "" (circle B.Cu 1750 -5080 0))<br>(keepout "" (circle F.Cu 1750 5080 0))<br>(keepout "" (circle B.Cu 1750 5080 0)) | (keepout "" (circle F.Cu 1750 -5080 0))<br>(keepout "" (circle B.Cu 1750 -5080 0))<br>(keepout "" (circle F.Cu 4000))<br>(keepout "" (circle B.Cu 4000))<br>(keepout "" (circle F.Cu 1750 5080 0))<br>(keepout "" (circle B.Cu 1750 5080 0))<br>(keepout "" (circle F.Cu 0 2540 5080))<br>(keepout "" (circle B.Cu 0 2540 5080))<br>(keepout "" (circle F.Cu 0 -3810 2540))<br>(keepout "" (circle B.Cu 0 -3810 2540))

I believe this is a change which is causing this issue: https://github.com/perigoso/keyswitch-kicad-library/pull/47

perigoso, 2023-02-22:
I'm guessing this is because of the custom through holes I implemented, in that the solder mask on top is smaller than on the bottom, I will need to take a look at what exactly is going on here

You can generate the footprints with regular through holes with a bit of tinkering for now, If needed

perigoso, 2023-05-21:
I reverted that change for v2.4, will revisit in the future with more consideration

closed 2023-05-21 by perigoso