kiswitch/kiswitch #59: SW_Hotswap_Kailh_MX_plated DRC errors

rukai, 2023-03-07:
I'm pretty sure these only started appearing as errors since kicad 7.0, but maybe I just hadnt noticed them before.

There appears to be a second smaller pad placed on top of the actual pad, can we just remove the second pad to fix this?
[image]

Probably isnt anything that can be done for this, seems to be caused by the small stabilizing hole placed near the large stabilizing hole:
[image]

Anathae, 2023-09-21:
I'm not sure exactly what this means, but I did some looking at the foot print in the foot print editor, and it appears that the issue lies with the fact that there are SMD Aperture pads defining where the solder paste and solder masks are for this foot print, and with the changes between KiCAD 6.99 and 7.00, this is getting flagged as an error.  As this looks like a completely valid way of constructing a foot print for this particular device, this may be a bug in the DRC rules in KiCAD.

LoaderB0T, 2024-02-26:
I am having the same issue with KiCAD 7, I am also sure that I did not have those problems in the past when I was probably still using KiCAD 6.
[image]