kiswitch/kiswitch #69: Oval stabilizer hole for Kailh choc V2

cheyao, 2024-10-08:
Hello!

I'm making a macropad with kailh choc v2s, and I've been matching your library with the kailh datasheets, but I've noticed that in previous issues discussing this switch you were using round holes for the stabilizer pin (Not sure if it's called like this):

[image]

But I've noticed that on the [datasheet](https://cdn.shopify.com/s/files/1/0657/6075/5954/files/SPEC-CPG135301D01_Kailh_Choc_V2_Low_Profile_Red_Switch_03cd29a8-065e-45f8-8053-b72dcd70616d.pdf?v=1667190153) (acquired from [here](https://www.kailh.net/products/kailh-choc-v2-low-profile-switch-set)) Uses a oval hole for the stabilizer pin on the bottom left.

[image]

Is this some kind of mistake in the library or a mistake on my part?

Thanks.

reinhrst, 2024-12-08:
I saw your comment, so I adjusted the holes on my own design to be oval. When I got the boards back, the holes were not drilled. [I understand now why not](https://forum.kicad.info/t/oval-mounting-holes/31294/3). 🤦 

I guess for this reason it's also better not to use oval holes in the footprints. I'm not sure what would have happened if I had actually designed the part to have 2*1.5mm hole rather than an oval hole.

In the end I don't think it matters much, the round hole seems to be in the right spot for the pin (I will not reorder the board, just do some Dremel magic).

cheyao, 2024-12-08:
Oh no! Sorry for the false information >.<

Now I am worrying about how my boards will come out, I think I also used oval holes for the holes >.<

Hope that your keypad was a success!

closed 2024-12-08 by cheyao

reinhrst, 2024-12-11:
The information was not false... it was very useful. There just turns out to be a practical reason not to have it.

As for extra information: I also read that PCD manufacturers don't allow two holes that are partly overlapping, as it's a good way to destroy your drill bits.