kiswitch/kiswitch #71: Library is installed through plugin manager, but I can't use the components

Neiizo, 2024-12-20:
I am very new to kicad, so sorry in advance.
I wanted to make a custom pcb, so I downloaded the plugin through the plugin manager.
I've made sure, that the path were added,
[image]
 and they are.
 However, as soon as I go into my project, there aren't any paths anymore. So I've tried adding it manually, but it didn't work, saying it couldn't be found
[image]
 Do you know why this is happening?
 
 Thanks in advance

perigoso, 2024-12-24:
Hi, Sorry I've been neglecting this so I'm not on top of the current issues. I'm not totally sure but it looks like we use some references to a KiCad 6 path variable and you are using KiCad 8.

This needs to be updated on the library itself, it's not your fault, but as a workaround you might be able to fix his by changing `${KICAD6_3RD_PARTY}` when you see it

humainbyte, 2025-01-22:
Just replace ${KICAD6_3RD_PARTY} with ${KICAD8_3RD_PARTY} and that should work. 

In your first screenshot, you can see that KiCAD is proposing paths, and the path for third party is ${KICAD8_3RD_PARTY}.